SAP/e-mobility-charging-stations-simulator #36: [FEATURE] Implement missing OCPP 1.6 commands

jerome-benoit, 2022-06-08:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

See README.md for the list.

jerome-benoit, 2022-08-28:
The firmware update is the most awaited command: 
+ handle an archive type like tar
+ define a JSON file format to simulate the target firmware version and update behavior, like
```json
{
	"firmwareVersion": "",
	"reset": true/false
	"resetDelay": "" 
}
```
+ implement the OCPP command with the archive download from FTP 
+ implement the firmware update simulation

jerome-benoit, 2022-09-11:
The second most awaited command is get charging profile.

jerome-benoit, 2023-05-27:
PR https://github.com/SAP/e-mobility-charging-stations-simulator/pull/574 implements:
+ `ReserveNow`
+ `CancelReservation`

syuusei3, 2024-05-30:
The first OCPP 2 command to implement is "GetBaseReport". 
 
To do it: 
- Mock OCPP 2 server with the mobility house framework
- Document the OCPP 2 mock building 
- Add request/response payload types definition
- Identify the payload validation JSON schemas
- Add the payload validation JSON schemas in the proper OCPP payload validation hooks
- Identify the file module to where the command implementation should be added
- Implement and hook the command code in the OCPP 2 code
- Test with the mock OCPP 2 server