SAP/e-mobility-charging-stations-simulator #39: [FEATURE] Add OCPP 2.X.X support

jerome-benoit, 2022-06-09:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

Per request/response modules.
Required for the Mobena project: [vedecom.fr/initiative-mobena](https://www.vedecom.fr/initiative-mobena/)

syuusei3, 2024-06-03:
The first OCPP 2 command to implement is "GetBaseReport".

To do it:

Mock OCPP 2 server with the mobility house framework
Document the OCPP 2 mock building
Add request/response payload types definition
Identify the payload validation JSON schemas
Add the payload validation JSON schemas in the proper OCPP payload validation hooks
Identify the file module to where the command implementation should be added
Implement and hook the command code in the OCPP 2 code
Test with the mock OCPP 2 server

jerome-benoit, 2024-06-03:
The first missing OCPP 2 command to implement is GetBaseReport. 
 
To do it: 

- Mock OCPP 2 server with the mobility house framework
- Document the OCPP 2 mock building 
- Add request/response payload types definition
- Identify the payload validation JSON schemas
- Add the payload validation JSON schemas in the proper OCPP payload validation hooks
- Identify the file module to where the command implementation should be added
- Implement and hook the command code in the OCPP 2 code
- Test with the mock OCPP 2 server