SAP/e-mobility-charging-stations-simulator #55: [FEATURE] Vue js web ui implement protocol to communicate with the simulator

HazbinFaulted, 2022-06-15:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

## List simulated charging stations ##
1. Send `listChargingStations` command through websocket
2. While waiting, use any mean (loading animation...) to notify user the request is being processed
3. Once response received, display it

## Start transaction command ##
1. Send `startTransaction` command through websocket
2. While waiting, use any mean (loading animation...) to notify user the request is being processed
4. Once response received, display it

## Stop transaction command ##
1. Send `stopTransaction` command through websocket
2. While waiting, use any mean (loading animation...) to notify user the request is being processed
5. Once response received, display it

jerome-benoit, 2022-08-24:
The Web UI has initial support for listing charging stations and start transaction. 
Support for: 
+ stopTransaction
+ startChargingStation
+ stopChargingStation
+ openConnection
+ closeConnection
is still missing

The Web UI design can also be improved.

jerome-benoit, 2022-08-26:
Since commit https://github.com/SAP/e-mobility-charging-stations-simulator/commit/16fe3949f743c57db407cc02a5c017af4da1c7a9, all supported UI protocol commands are implemented.

closed 2022-08-26 by jerome-benoit