SAP/e-mobility-charging-stations-simulator #69: [FEATURE] Installation instructions

hughsheehy, 2022-06-27:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

Working through the installation instructions, it's not 100% clear which config files I should be copying/editing/using. 

I've copied the config-template.json file in /src/assets into a config.json file, with a change in the URL for the OCPP server from localhost to the URL of an actual OCPP server. Should I also be copying or changing other template files? It's not clear. 

When I try to npm start after doing that (windows 10), I get the following outputs. 
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

PS D:\hughs\e-mobility-charging-stations-simulator-main> npm start
npm WARN config global `--global`, `--local` are deprecated. Use `--location=global` instead.

> e-mobility-charging-stations-simulator@1.1.63 prestart
> npm run build

npm WARN config global `--global`, `--local` are deprecated. Use `--location=global` instead.

> e-mobility-charging-stations-simulator@1.1.63 build
> npm run rollup

npm WARN config global `--global`, `--local` are deprecated. Use `--location=global` instead.

> e-mobility-charging-stations-simulator@1.1.63 rollup
> rollup --config

'rollup' is not recognized as an internal or external command,
operable program or batch file.
PS D:\hughs\e-mobility-charging-stations-simulator-main>

jerome-benoit, 2022-06-27:
You first have to run npm install. It's not documented because it's a implicit when you want to run node.js code.

hughsheehy, 2022-06-27:
> npm install

I try to follow the exact instructions. Jumping from language to language means that I don't know what should be implicit.

jerome-benoit, 2022-06-27:
A PR to enhance the README will be welcome

closed 2022-07-01 by jerome-benoit

hughsheehy, 2022-07-03:
I'm very reluctant to do PRs on well run projects. 

But another question on the readme. Should the "npm start:server" instruction be "npm run start:server"

H

commit 007ee642 (2022-07-01): Add more detailled installation instructions

Close #69