SAP/e-mobility-charging-stations-simulator #333: [HELP]

PradeepGanapathi, 2022-12-28:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

- after cloning the code I have given npm install on root directory.
- then npm start to start the server. 
- started the client side server also.
- after clicking the start simulator button on client side in console it shows the error **WebSocket connection to 'ws://localhost:8080/' failed:**

### e-mobility-charging-stations-simulator version

1.1.88

### Node.js Version

v16.17.1

### System

 OS: macOS 12.5
    CPU: (8) arm64 Apple M1
    Memory: 91.77 MB / 8.00 GB
    Shell: 5.8.1 - /bin/zsh

### Expected result

Expected to start the webSocket connection and able to view the list of charging stations after clicking the start simulator.

### Actual result

Actually it shows the error like WebSocket connection to 'ws://localhost:8080/' failed:

### Attachments

[image]

jerome-benoit, 2022-12-28:
You need do not have an OCPP-J server to connect to (here at ws://localhost:8080/).

robohydra in the repo can be used as WS server. But you then will have to manually do the OCPP-J message sequence on robohydra message endpoint.

PradeepGanapathi, 2022-12-29:
[image]

but after giving npm start in root folder it shows simulator started.

jerome-benoit, 2022-12-29:
Sorry, I confounded WebSockets server. You need to enable the UI server in the main configuration file with authentication enabled: 
 
```json
"uiServer": {
    "enabled": true,
    "type": "ws",
    "authentication": {
      "enabled": false,
      "type": "basic-auth",
      "username": "admin",
      "password": "admin"
    }
  },
```

PradeepGanapathi, 2022-12-29:
Thanks for your kind response.
- I have changed the configuration as per your instruction.
[image]

- I'm still getting the same error.
[image]

jerome-benoit, 2022-12-29:
> [image]

Please follow the simulator README.md and read it thoroughly: the configuration file template must be copied to config.json.

PradeepGanapathi, 2022-12-29:
Thank you so much, it worked.

closed 2022-12-29 by jerome-benoit