SAP/e-mobility-charging-stations-simulator #524: [BUG]

Pr3c10us, 2023-05-10:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

when i run pnpm start i get this error
Error: ENOENT: no such file or directory, open '/e-mobility-charging-stations-simulator-main/performance Records.json' 

### e-mobility-charging-stations-simulator version

latest

### Node.js Version

18

jerome-benoit, 2023-05-10:
If you disable the performance storage in the configuration file, that should go away until the bug is fixed:  

```json
performanceStorage: {
"enabled": false,
"type": "jsonfile",
"file:///performanceRecords.json"
}
```

closed 2023-05-10 by jerome-benoit

Pr3c10us, 2023-05-10:
> If you disable the performance storage in the configuration file, that should go away until the bug is confirmed and fixed:
> 
> ```json
> "performanceStorage": {
>   "enabled": false,
>   "type": "jsonfile",
>   "file:///performanceRecords.json"
> },
> ```

> 

Thanks it worked

commit f682b2dc (2023-05-10): fix: handle relative path to performance records JSON file

fix #524