SAP/e-mobility-charging-stations-simulator #553: [BUG]

Pr3c10us, 2023-05-17:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

After Importing the http ui-protocol to insomnia i can't seem to send request. Request doesn't go through just timeout. I have tried searching for cause or solution but to no avail. Is there any requirement before i can start sending request with insomnia or what else can i do

### e-mobility-charging-stations-simulator version

latest

### Node.js Version

18

jerome-benoit, 2023-05-17:
The same requirements as for the Web UI: https://github.com/SAP/e-mobility-charging-stations-simulator/tree/main/ui/web. 

The WebSocket requests collection must then be used.

Pr3c10us, 2023-05-17:
> 

I did that still not working
"uiServer": {
    "enabled": true,
    "type": "ws",
    "authentication": {
      "enabled": false,
      "type": "basic-auth",
      "username": "admin",
      "password": "admin"
    }
  },

jerome-benoit, 2023-05-17:
> > After Importing the http ui-protocol to insomnia 
>
> I did that still not working 
> 

"The WebSocket requests collection must then be used."

jerome-benoit, 2023-05-17:
> > "The WebSocket requests collection must then be used."
> 
> connect ECONNREFUSED 127.0.0.1:8080 error when i try to connect

If `netstat -patn` (or the similar command for your OS) says that a node server is listening on port 8080, it's then an issue related to your OS or insomnia setup and totally unrelated to the simulator code. 

Please stop using GitHub issues that are dedicated to simulator development for its usage support. Ask a forum on your OS, or insomnia. 

It's the last time I tell it, next time a BUG that is not a BUG is opened, I will ignore it and at some point in time, close it. 

Thanks.

closed 2023-05-26 by jerome-benoit