SAP/e-mobility-charging-stations-simulator #582: [BUG] SyntaxError: Named export 'Queue' not found. The requested module 'mnemonist' is a CommonJS module

hughsheehy, 2023-05-30:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

On start, with pnpm start, I get a lot of output, and in the middle there's this message. 

SyntaxError: Named export 'Queue' not found. The requested module 'mnemonist' is a CommonJS module, which may not support all module.exports as named exports.
CommonJS modules can always be imported via the default export, for example using:

import pkg from 'mnemonist';

### e-mobility-charging-stations-simulator version

1.2.14

### Node.js Version

16

### System

System:
    OS: Linux 5.10 Debian GNU/Linux 11 (bullseye) 11 (bullseye)
    CPU: (2) x64 AMD EPYC 7B12
    Memory: 564.53 MB / 975.63 MB
    Container: Yes
    Shell: 5.1.4 - /bin/bash

### Expected result

The program to start. 

### Actual result

Error observed.

hughsheehy, 2023-05-30:
Further down, the ending error is this. 

    at ModuleJob._instantiate (node:internal/modules/esm/module_job:123:21)
    at ModuleJob.run (node:internal/modules/esm/module_job:189:5)
    at async Promise.all (index 0)
    at ESMLoader.import (node:internal/modules/esm/loader:530:24)
    at loadESM (node:internal/process/esm_loader:91:5)
    at handleMainPromise (node:internal/modules/run_main:65:12)
 ELIFECYCLE  Command failed with exit code 1.

jerome-benoit, 2023-05-30:
If the error is only related to the `pnpm build` command, then the issue is related to your local node.js setup. The CI builds for node.js 16 is successful. 
If not, you can try 

`import Queue from 'mnemonist/queue.js'`

hughsheehy, 2023-05-30:
that import - is it in 

https://github.com/SAP/e-mobility-charging-stations-simulator/blob/13aa228baf773556a558eae4b6aa6170d57941b7/src/utils/AsyncLock.ts#L4

Not sure what could be wrong with my node.js setup. It's a fresh install on a fresh server. Any ideas/suggestions?

H

jerome-benoit, 2023-05-30:
> Not sure what could be wrong with my node.js setup. It's a fresh install on a fresh server. Any ideas/suggestions?

I just gave you the import syntax to test in the previous comment ... 

Probably also: 

`import LRUCache from 'mnemonist/lru-map-with-delete.js'` in other file.

jerome-benoit, 2023-05-30:
Could you please confirm or infirm that the import syntax change makes the simulator works with node.js 16.x?

hughsheehy, 2023-05-30:
Hi
I'm not sure what's going on. I started again and downloaded the last release as zipped source code and tried that - instead of doing git clone. 

The import queue error does not happen because that line in not in the AsyncLock.ts file in the release file. 

I made a video of the install process. See [here](https://www.youtube.com/watch?v=IxdjJyowpEg)

The install works, though there are warnings, and on pnpm start the server starts. But then it stops after about 90 seconds. 

See screenshots. 
[image]
[image]

jerome-benoit, 2023-05-30:
The README.md explicitly state that idtags.json must be created from its template. 

And as a rule: first test locally and once it's working, go for a remote install.

closed 2023-05-30 by jerome-benoit

hughsheehy, 2023-06-14:
I edited the idtags.json as you suggested. 

The server starts, but never connects to anything. THe old software ran perfectly. And is running perfectly. So I think I'll leave this one.

Oh, on the install instructions, you might consider adding all the missing bits like needing to install husky (and how) and having to install corepack, etc.  We're not all into Node.

jerome-benoit, 2023-06-14:
> 
> Oh, on the install instructions, you might consider adding all the missing bits like needing to install husky (and how) and having to install corepack, etc. We're not all into Node.

The install instructions in the README.md already contains all that bits. Would you mind to read it?

jerome-benoit, 2023-06-14:
> The server starts, but never connects to anything. THe old software ran perfectly. And is running perfectly. So I think I'll leave this one.

It's a configuration issue on your side in `config.json` or charging station templates. And it's out of the scope of the GitHub issues coverage which are dedicated to development, not end-user support on how OCPP works, how to configure an OCPP server URL, how to use Insomnia, etc. for obvious lack of bandwidth available to do it.

commit f31d1d0c (2023-05-30): fix: fix commonjs import issue with node.js 16.x.x

Close #582