SAP/e-mobility-charging-stations-simulator #639: [BUG] Can't start the simulator following the README

abhishandilya, 2023-07-26:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

I can't seem to run the simulator using the README

I have setup Node and PNPM, installed the deps, setup config but when I run the `pnpm start`, I don't see helpful logs or the URL to the Web UI.

Please guide me.

### e-mobility-charging-stations-simulator version

1.2.19

### Node.js Version

16.20.1

### System

OS: macOS 13.5
CPU: (8) arm64 Apple M1 Pro
Memory: 149.52 MB / 16.00 GB
Shell: 5.9 - /bin/zsh

### Actual result

```sh
....
created dist in 2.9s
Charging stations simulator 1.2.19 started with 10 charging station(s) from 5 configured charging station template(s) and 10 worker(s) concurrently running in 'workerSet' mode (1 charging station(s) per worker)
Worker set/pool information: {
  version: '1.0.0',
  type: 'set',
  worker: 'thread',
  size: 10,
  elementsExecuting: 10,
  elementsPerWorker: 1
}
```

abhishandilya, 2023-07-26:
I have a OCPP server running on AWS and use its URL on the `supervisionUrls`

jerome-benoit, 2023-07-26:
> ```shell
> Charging stations simulator 1.2.19 started with 10 charging station(s) from 5 configured charging station template(s) and 10 worker(s) concurrently running in 'workerSet' mode (1 charging station(s) per worker)
> ```

Please do not open bug report that report no bug.  
That project does not have the bandwidth to do end users support via GitHub issues. 
GitHub issues are dedicated to development tasks, not simulator usage support. And the configuration tunables are self-explanatory enough to allow advanced end user to debug themselves their configuration simulator issues via the logs. 
I will probably add a big fat warning at GitHub issues creation to explain the project GitHub issues purpose.

closed 2023-07-27 by abhishandilya

abhishandilya, 2023-07-27:
Thank you for the kind reply.

CleberPeter, 2024-01-17:
> in the project GitHub issues purpose.

Hi @abhishandilya you have to run:

* pnpm install 
* pnpm start 

on ui/web folder.