SAP/e-mobility-charging-stations-simulator #704: [BUG] - Issue with logPrefix()

ebram-tw, 2023-09-08:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

Hi!  I get the following error:  `TypeError: Kr.logPrefix is not a function` in my compiled code with the somewhat helpful stackstrace:

```
TypeError: Kr.logPrefix is not a function
    at Kr.warnDeprecatedConfigurationKey (file:///code/ev-simulator/dist/Bootstrap-69df843b.js:1:43201)
    at Kr.buildLogSection (file:///code/ev-simulator/dist/Bootstrap-69df843b.js:1:37975)
    at <static_initializer> (file:///code/ev-simulator/dist/Bootstrap-69df843b.js:1:33833)
    at file:///code/ev-simulator/dist/Bootstrap-69df843b.js:1:33362
    at ModuleJob.run (node:internal/modules/esm/module_job:194:25)
```

I noticed that the definition of `logPrefix` is different between Utils.ts and Configuration.ts:
https://github.com/SAP/e-mobility-charging-stations-simulator/blob/main/src/utils/Utils.ts#L19-L21
https://github.com/SAP/e-mobility-charging-stations-simulator/blob/main/src/utils/Configuration.ts#L290-L292

I believe this is something wrong with the declaration in Configuration.ts--I'm wondering if that was supposed to be a `const` as well.  

### e-mobility-charging-stations-simulator version

1.2.20

jerome-benoit, 2023-09-08:
Could you please start the development build to have a stacktrace with source mapping enabled?

ebram-tw, 2023-09-08:
Hi @jerome-benoit thanks for responding so quickly.  I will try to find some time to do that--for now I just removed the code in `Configuration.ts` referencing `logPrefix` to get around the issue.

jerome-benoit, 2023-09-09:
> Hi @jerome-benoit [...]

Please fill all the fields in the issue template. Each one is there for a good reason.

And retest with the current node.js LTS latest version or latest stable.

closed 2023-09-09 by ebram-tw