SAP/e-mobility-charging-stations-simulator #711: [BUG]

samarasdelta, 2023-09-12:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

After installing the new version, I came across this bug

```
Startup error:  Error: ENOENT: no such file or directory, mkdir '/C:/wamp64/www/e-mobility-charging-stations-simulator-1.2.21/performance'
    at mkdirSync (node:fs:1349:3)
    at Vn.open (file:///C:/wamp64/www/e-mobility-charging-stations-simulator-1.2.21/dist/Bootstrap-c4b7dece.js:1:78365)
    at zn.start (file:///C:/wamp64/www/e-mobility-charging-stations-simulator-1.2.21/dist/Bootstrap-c4b7dece.js:1:81842)
    at async file:///C:/wamp64/www/e-mobility-charging-stations-simulator-1.2.21/dist/start.js:1:554 {
  errno: -4058,
  syscall: 'mkdir',
  code: 'ENOENT',
  path: '/C:/wamp64/www/e-mobility-charging-stations-simulator-1.2.21/performance'
}
```

I made a folder named "performance" as the error description dictates, but it didn't seem to work.
Also made a file with the same name etc.

Any reply would be appreciated.

### e-mobility-charging-stations-simulator version

1.2.21

### Node.js version

16.15.0

### System

  System:
    OS: Windows 10 10.0.22621
    CPU: (16) x64 AMD Ryzen 7 5700G with Radeon Graphics

jerome-benoit, 2023-09-12:
> Node.js version
> 
> 16.15.0
> 
> 
> 
Please the current node.js LTS version. The simulator does not work with the previous node.js LTS version. The version check is now enforced in the main branch.

samarasdelta, 2023-09-12:
Solved this after adding the section bellow in the config.js file. Actually it was missing the "file" entry by default.

```
  "performanceStorage": {
    "enabled": false,
    "type": "jsonfile",
    "file": "performanceRecords.json"
  },
```

closed 2023-09-12 by samarasdelta