SAP/e-mobility-charging-stations-simulator #916: SupportedFeatureProfiles configuration not applied[BUG]

aidnani1, 2024-01-03:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

I updated the SupportFeatureProfiles in the station configuration file. I added SmartCharging to the list, but when the configuration is sent to the CSMS, the SupportFeatureProfiles defaults to Core,FirmwareManagement,LocalAuthListManagement. Is SmartCharging not supported? 

I also tried to force it my making a temporary code change, but it doesn't look the MeterValues message take the charge limit from TxProfile into account.

### e-mobility-charging-stations-simulator version

1.0.1

### Node.js version

21.4.0

### System

  System:
    OS: macOS 14.0
    CPU: (8) arm64 Apple M2
    Memory: 98.34 MB / 16.00 GB
    Shell: 5.9 - /bin/zsh

### Expected result

SmartCharging to be included in the SupportFeatureProfiles and MeterValues to reflect the limits from TxProfile

### Actual result

SmartCharging not included in SupportedFeatureProfiles. MeterValues continue to report random values.

### Steps to reproduce

Change SupportedFeatureProfiles in station config. Observe the changes do not take effect.

### Attachments

None

jerome-benoit, 2024-01-04:
More informations is needed such as the: 
+ charging station template file
+ charging station configuration file
+ charging station simulator configuration file

aidnani1, 2024-01-04:
Simulator version 1.2.31. The simulator configuration file and the charging station template file are attached.
[config.json](https://github.com/SAP/e-mobility-charging-stations-simulator/files/13832420/config.json)
[siemens.station-template.json](https://github.com/SAP/e-mobility-charging-stations-simulator/files/13832421/siemens.station-template.json)

Sorry for providing the wrong simulator version before. Let me know if you need anything else.

jerome-benoit, 2024-01-04:
The OCPP configuration section is not expected to be taken into account after the initial charging station configuration file is created from the template. 
Please read the documentation on configuration files hierarchy and priority.

aidnani1, 2024-01-04:
Isn't that file overwritten when I do a pnpm start after changing the template file? I always do a pnpm start after making any change to src/assets/config.json or src/assets/station-templates/siemens.station-template.json.

jerome-benoit, 2024-01-04:
No. Please read the README that explain configuration files hierarchy and priority.

jerome-benoit, 2024-01-05:
Closed as not a bug: the OCPP configuration must be changed directly in the charging station configuration file, not in the template for already existing charging stations on the simulator.

closed 2024-01-05 by jerome-benoit