SAP/e-mobility-charging-stations-simulator #959: [BUG]  RemoteStartTransaction reply is sent only after StatusNotification.req + Authorize.req + StartTransaction.req

Ohad31415, 2024-01-29:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

`handleRequestRemoteStartTransaction` is waiting only after status-notification + authorize if needed + start-transaction to send the central-system a reply on the remote-transaction, isn't it not according the the protocol/docs?

### e-mobility-charging-stations-simulator version

1.0.1

### Node.js version

v21.5.0

### System

  System:
    OS: macOS 14.3
    CPU: (10) arm64 Apple M1 Pro
    Memory: 77.25 MB / 16.00 GB
    Shell: 5.9 - /bin/zsh

### Expected result

The simulator replying with `RemoteStartTransaction.conf` and then manage the sequence of `StatusNotification`, `Authorize` and `StartTransaction`

jerome-benoit, 2024-01-29:
Could you please extract the OCPP spec content that seems not respected by the code in the repo?
Are you sure about the simulator version reported in the issue?

Ohad31415, 2024-01-29:
Thanks for the reply!
According to the docs here https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pd -
RemoteStartTransaction is no different then other message that each request should get a reply result/error, the remote-start if accepted has effects later that it shall do+send status, authorize if needed, update and start-transaction

[image]

jerome-benoit, 2024-01-29:
Where exactly the code miss to follow the spec? The code behave as written, expect if a template tunable flag is changed.

Ohad31415, 2024-01-29:
What flag is it? maybe I missed it

jerome-benoit, 2024-01-29:
The start transaction response is checked only if `AuthorizeRemoteTxRequests` is set to `true`, which is not the behavior described in the specs.

Ohad31415, 2024-01-29:
That's fine and following the docs completely, but why the RemoteStartTransaction reply isn't sent after the central-system request and is only sent after the start-transaction reply from central-system?  

I think that this is the right flow - 

```mermaid
sequenceDiagram
autonumber

CS->>CP: RemoteStartTransaction.req
CP-->>CS: RemoteStartTransaction.conf

CP->CS: Should authorize?
CP->CS: StatusNotification

CP->>CS: StartTransaction.req
CS-->>CP: StartTransaction.conf
```

Where currently it sends the central-system the `RemoteStartTransaction.conf` after the `StartTransaction.conf` instead of after the `RemoteStartTransaction.req`

jerome-benoit, 2024-01-29:
The asynchronous authorization check described in the spec is: 
+ pointless
+ require some supplemental code to stop the transaction if finally the start transaction response is not accepted

I do not see the point to start the transaction if `AuthorizeRemoteTxRequests` is set to false, then check if the transaction can be started via a `startTransaction`, then conditionally stop the started transaction ... 
If I only make the code return an accepted answer without adding the `startTransaction` check and conditional `stopTransaction`. 
So I will implement a synchronous authorization check for all cases.

jerome-benoit, 2024-01-29:
@Ohad31415: Could you please confirm or infirm that PR fix the issue by checking out its branch?

Ohad31415, 2024-01-29:
I still doesn't reply to the `RemoteStartTransaction` first and does it after waiting for response for `StartTransaction` from central-system

jerome-benoit, 2024-01-29:
@Ohad31415: and with the latest commit in the commit triggering a delayed start transaction?

Ohad31415, 2024-01-29:
I still get the start-transaction before the reply on the remote-start-transaction

jerome-benoit, 2024-01-29:
The event emission is queued in the promises queue. Hopefully the FIFO will queue it after the OCPP request promise resolving in most case.

closed 2024-01-29 by jerome-benoit

jerome-benoit, 2024-01-30:
> The event emission is queued in the promises queue. Hopefully the FIFO will queue it after the OCPP request handler promise resolving in most case.

The latest code in main implement a deterministic chronologically ordered 2 stages incoming request handling framework.

PR #960 (2024-01-29): fix: fix control flow at remote start
closes #959
commit b10202d7 (2024-01-29): fix: fix control flow at remote start

closes #959