SAP/e-mobility-charging-stations-simulator #964: [FEATURE] Add a create station in the UI protocol as API

OS-SAP, 2024-01-30:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

We need to create remotely a set of stations configuration on a simulator running instance.
Would it be possible to add this feature ?

Need to investigate how to pass con and identify templates maybe ... 
 
https://github.com/SAP/e-mobility-charging-stations-simulator/blob/main/README.md#ui-protocol

thx

jerome-benoit, 2024-01-30:
1. Add endpoint on UI service to list the templates availables  (files basename)
2. Add endpoint on UI service to add n charging chargings from template X: 
```ts
{
template: string
number: number
start?: boolean
} 
```
3. Add the template base name to the charging stations list endpoint 
4. The number of started station at simulator startup remains the one hardcoded in main `config.json` file.  

PR welcome.

closed 2024-02-01 by jerome-benoit

stephanedupin, 2024-02-01:
Thanks for the quick implementation. We are going to test it, in the meanwhile is it possible to clarify the following aspects in relation to what has been changed:
- will the newly added stations be dispatched according to workers setup defined in the config.json? 
- in case of a simulator deployed in BTP, all stations configs will be destroyed when stopping/restarting the simulator? (not a problem in itself, only to clarify, we don't have a requirement for persistence)

jerome-benoit, 2024-02-01:
> Thanks for the quick implementation. We are going to test it, in the meanwhile is it possible to clarify the following aspects in relation to what has been changed:
> 
> * will the newly added stations be dispatched according to workers setup defined in the config.json?

Yes. So the configuration must be tuned for the total expected running charging stations *before*.  For example 'auto' will not do it for `workerSet`.

> * in case of a simulator deployed in BTP, all stations configs will be destroyed when stopping/restarting the simulator? (not a problem in itself, only to clarify, we don't have a requirement for persistence)

Charging station configurations are persistent unless configured otherwise. Manual deletion is the only way to get rid of them (app redeploy, destroy, ...). The number of started charging station at simulator startup will always be the ones in the main configuration files (that's the meaning of the sentence: "The number of started station at simulator startup remains the ones hardcoded in main config.json file after a stop/start lifecycle" in the specs below).

commit c5ecc04d (2024-02-01): feat: add addChargingStations command to UI API

closes #964