SAP/e-mobility-charging-stations-simulator #991: [BUG] UI does not work anymore on simulator v 1.2.37

aidnani1, 2024-02-23:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Description

I downloaded the latest version of code today. After the upgrade, I started UI server from ui/web on localhost:3030. But the browser shows a blank screen. I tried two different browsers - Chrome and Safari. Both show a blank page. I looked at the network traffic and I see the following files are returned by UI server
index-DSI7Tebn.js, index-BE4WkIGx.css and config.json

### e-mobility-charging-stations-simulator version

1.2.37

### Node.js version

v21.6.2

### System

  System:
    OS: macOS 14.2.1
    CPU: (8) arm64 Apple M2
    Memory: 80.27 MB / 16.00 GB
    Shell: 5.9 - /bin/zsh

### Expected result

The screen to show the preview that is shown on the READMe page for UI. It used to work until I upgraded to 1.2.37

### Actual result

I see a blank page on browser

### Steps to reproduce

- Start Charger simulator (pnpm start) from e-mobility-charging-stations-simulator/src dir
- Start UI/Web (pnpm start) from e-mobility-charging-stations-simulator/ui/web
- Using a browser navigate to http://localhost:3030/

Its quite possible i did something wrong. But I cannot tell what. I had tried earlier in the day with the older version of Sim I had and it had worked. I did a git pull and now it doesn't work anymore. :(

aidnani1, 2024-02-23:
Ignore this. This may be an issue on my side. I used a different terminal window to start the Charger Simulator and kept everything else the same and it worked. Not sure why.

jerome-benoit, 2024-02-24:
The configuration file path and format for the Web UI has changed. Please ensure you have follow the updated documentation.

jerome-benoit, 2024-02-24:
> * Start Charger simulator (pnpm start) from e-mobility-charging-stations-simulator/src dir

`pnpm start` is expected to be executed from the repository root directory.

jerome-benoit, 2024-02-25:
Closed because not a bug.

closed 2024-02-25 by jerome-benoit