SAP/e-mobility-charging-stations-simulator #1006: [BUG] GetDiagnostics is trying to upload the incorrect folder

nahueld, 2024-03-04:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

`handleRequestGetDiagnostics` is not pointing to the right directory, current code:
```
        const logFiles = readdirSync(resolve(dirname(fileURLToPath(import.meta.url)), '../'))
          .filter(file => file.endsWith('.log'))
          .map(file => join('./', file))
```

### Version

1.2.38

### Node.js version

18

### System

  System:
    OS: macOS 13.6.4
    CPU: (12) arm64 Apple M2 Pro
    Memory: 115.53 MB / 16.00 GB
    Shell: 5.9 - /opt/homebrew/Cellar/zsh/5.9/bin/zsh

### Expected result

The path should point to the `logs` folder and logs should be uploaded to the FTP server.

### Actual result

No log files are found  and this causes:
```
OCPPIncomingRequestService.handleIncomingRequestError: Incoming request command 'GetDiagnostics' error: no files or directories specified
```

### Steps to reproduce

Try to use GetDiagnostics

nahueld, 2024-03-04:
I already have a fix for this and I'm happy to push

jerome-benoit, 2024-03-05:
> I already have a fix for this and I'm happy to push, the fix is:

Thanks for both fixes. 

Please open PRs directly next time to propose the fix to ease its integration.

commit 0ddd2460 (2024-03-05): fix: get diagnostics is trying to upload the incorrect folder

use log configuration section to build the archive content

closes #1006

Contributed-by: @nahueld
closed 2024-03-05 by jerome-benoit