SAP/e-mobility-charging-stations-simulator #1034: [BUG] Double trailing slash when supervisionUrl points to domain path root

saschnet, 2024-04-24:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

When supervisionUrl points to a raw domain, e.g. `ws://127.0.0.1:8000`, the resulting requests to the webserver are initiated to `ws://127.0.0.1:8000//{baseName}` instead of `ws://127.0.0.1:8000/{baseName}`.

This is due to the false assumption that and URL property in JavaScript never ends with an trailing slash. But it does when the URL points to the root path.

Possible fix:
```ts
  public get wsConnectionUrl(): URL {
    return new URL(
      `${
        this.stationInfo?.supervisionUrlOcppConfiguration === true &&
        isNotEmptyString(this.stationInfo.supervisionUrlOcppKey) &&
        isNotEmptyString(getConfigurationKey(this, this.stationInfo.supervisionUrlOcppKey)?.value)
          ? getConfigurationKey(this, this.stationInfo.supervisionUrlOcppKey)?.value
          : this.configuredSupervisionUrl.href
      }${this.configuredSupervisionUrl.pathname !== '/' ? '/' : ''}${this.stationInfo?.chargingStationId}`
    )
  }
```

### Version

master

### Node.js version

v18.20.2

### System

Not necessary for this bug

### Expected result

Access `ws://127.0.0.1:8000/{baseName}`

### Actual result

Access `ws://127.0.0.1:8000//{baseName}`

### Steps to reproduce

Set in config.js
```json
"supervisionUrls": [
        "ws://127.0.0.1:8000"
   ]
```

jerome-benoit, 2024-04-25:
Is it actually causing any connection issue?

saschnet, 2024-04-25:
For well configured Webservers ws://127.0.0.1:8000//{baseName} != ws://127.0.0.1:8000/{baseName}, hence it is leading to failing connections.

commit 8fa6f565 (2024-04-25): fix: avoid duplicated slash in connection url

closes #1034
closed 2024-04-25 by jerome-benoit