SAP/e-mobility-charging-stations-simulator #1039: [BUG] When triggering message BootNotification and its returning request Accepted, the emulator does not start the sequence of messages

nahueld, 2024-05-29:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

When triggering message BootNotification and its returning request Accepted, the emulator does not start the sequence of messages

### Version

1.3.3

### Node.js version

18

### System

  System:
    OS: macOS 13.6.4
    CPU: (12) arm64 Apple M2 Pro
    Memory: 129.89 MB / 16.00 GB
    Shell: 5.9 - /opt/homebrew/Cellar/zsh/5.9/bin/zsh

### Expected result

When the BootNotification result of a Trigger BootNotification is Accepted I expect the subsequent calls to be executed from the emulator, for example, StatusNotification

### Actual result

- The emulator does nothing
- Calling Trigger StatusNotification causes an Exception in the emulator

### Steps to reproduce

- Start the emulator
- Configure the CSMS to respond "Pending" to the first BootNotification
- Make necessary config key changes or any other complementary request you need to do in between the first BootNotification and the subsequent
- Once done send a TriggerMessage with BootNotification
- Accept the BootNotification incoming request from the emulator in the CSMS
- Notice that the StatusNotification requests do not arrive
- Notice that sending Trigger StatusNotification also fails

### Attachments

Potential fix I'm using on my branch:

File: `OCPP16IncomingRequestService.ts`
```
 this.on(
      OCPP16IncomingRequestCommand.TRIGGER_MESSAGE,
      (
        chargingStation: ChargingStation,
        request: OCPP16TriggerMessageRequest,
        response: OCPP16TriggerMessageResponse
      ) => {
        if (response.status !== OCPP16TriggerMessageStatus.ACCEPTED) {
          return
        }
        const { requestedMessage, connectorId } = request
        const errorHandler = (error: unknown): void => {
          logger.error(
            `${chargingStation.logPrefix()} ${moduleName}.constructor: Trigger ${requestedMessage} error:`,
            error
          )
        }
        switch (requestedMessage) {
          case OCPP16MessageTrigger.BootNotification:
            chargingStation.ocppRequestService
              .requestHandler<OCPP16BootNotificationRequest, OCPP16BootNotificationResponse>(
              chargingStation,
              OCPP16RequestCommand.BOOT_NOTIFICATION,
              chargingStation.bootNotificationRequest as OCPP16BootNotificationRequest,
              { skipBufferingOnError: true, triggerMessage: true }
            )
              .then(response => {
               //assign the boot notification response
                chargingStation.bootNotificationResponse = response
        
               //depending on the status trigger the respective event
                switch (response.status) {
                  case RegistrationStatusEnumType.ACCEPTED:
                    chargingStation.emit(ChargingStationEvents.accepted)
                    break
                  case RegistrationStatusEnumType.REJECTED:
                    chargingStation.emit(ChargingStationEvents.rejected)
                    break
                  default:
                }
              })
              .catch(errorHandler)
            break
```

jerome-benoit, 2024-05-29:
The CS registration status on the CSMS is not 'Accepted' before the OCPP command trigger?

nahueld, 2024-05-29:
Yes, it is not accepted, it is in fact "Pending", according to OCPP :

> The Central System MAY also return a Pending registration status to indicate that it wants to retrieve or set certain > > information on the Charge Point before the Central System will accept the Charge Point. If the Central System returns the Pending status, the communication channel SHOULD NOT be closed by either the Charge Point or the Central System. The Central System MAY send request messages to retrieve information from the Charge Point or change its configuration. The Charge Point SHOULD respond to these messages. The Charge Point SHALL NOT send request messages to the Central System unless it has been instructed by the Central System to do so with a TriggerMessage.req request.

We do:
1. Send Pending on initial BootNotification
2. We change some config keys
3. We request a new BootNotification through TriggerMessage
4. On the second BootNotification we return Accepted

jerome-benoit, 2024-05-29:
I think the fix you've made should be moved into the 'BootNotificationResponse' handler. There's something wrong in there since 'startMessageSequence()' is not triggered by the event 'ChargingStationEvents.accepted'.

jerome-benoit, 2024-05-29:
The CS event(s) emission code is only run at connexion opening, which is wrong. 
A proper fix is to either move that into the BootNotificationResponse request handler; or a generic OCPP command event handler for boot notification command.

jerome-benoit, 2024-05-29:
Please reopen if needed

commit 50539084 (2024-05-29): fix: ensure message sequence is fired at boot notification response
processing

closes #1039
closed 2024-05-29 by jerome-benoit