SAP/e-mobility-charging-stations-simulator #1040: [BUG] web-ui doesn't connect to simulator when running as container

Heitikol, 2024-06-02:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Web UI

### Description

When starting the simulator via docker compose up, the web-ui can't connect to the simulator.

### Version

1.3.3

### Node.js version

v20.12.2

### System

Win 10

### Expected result

Running `docker compose up` results in the calibrated charging stations being accessible via the UI. We expect the same behaviour as when running the simulator via  `pnpm start ` and  `pnpm preview` (i.e. not in a container)

### Actual result

Running `docker compose up` results in the configured charging stations successfully being simulated, e.g. trying to connect to the provided supervision URL maintained in the config-file in the docker-folder. 
But when opening the UI via http://localhost:3030/ it throws an error:

Toast:
1. Error: Error in WebSocket to UI server 'localhost'
2. Info: WebSocket to UI server closed

Browser console:
1. WebSocket connection to `ws://localhost:8080/` failed'
2. Error in WebSocket to UI server `localhost`

When running the simulator via `pnpm start ` and  `pnpm preview` the UI works as expected

### Steps to reproduce

1. clone repository
2. create config.json and idtags.json in src/assets as copy of the template files config-template.json and idtags.json
3. `cd ui`
4. `cd web`
5. run `docker compose up -d --build --force-recreate`
6. open http://localhost:3030/ with a browser of choice
7. notice that none of the configured stations are available

### Attachments

[image]
[image]

jerome-benoit, 2024-06-02:
- The configuration files used in the built docker image are not the ones in src/assets, but the ones in docker/ directory.
- The default configuration files in docker/ are probably buggy and not really tested since a long time.
- The simulator UI integration in the docker image is also not really tested and not known to work. 
- The simulator integration in the docker image is known to work with proper configuration. 

We do not have the bandwidth needed to maintain the docker image working. But PR are accepted ;)

jerome-benoit, 2024-06-04:
> We'll think about integrating the UI and simulator in a docker image. If we do, we'll provide a PR 

On a different docker image than the one running the simulator? 

Since the Web UI is far from being secure, the recommended setup is to deploy the simulator alone with the UI server enabled and protected behind a WSS reverse proxy. 
And setup the UI locally to connect to the multiples simulators instance deployed to monitor them. The authentication less UI is not exposed to the public.

commit 20fb1094 (2024-09-08): refactor: separate out dashboard docker image

closes #1040
closed 2024-09-08 by jerome-benoit