SAP/e-mobility-charging-stations-simulator #1055: [BUG] elementAddDelay should be elementStartDelay in the readme

bramley-jetcharge, 2024-06-19:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

Just a minor one, but it was difficult to spot.
The readme documents worker startup config `elementAddDelay`, but the field is actually called `elementStartDelay`.

This field is useful to add a delay between each charger connecting, so I don't DoS my ocpp server :)

### Version

latest

### Node.js version

not relevant

### System

not relevant

jerome-benoit, 2024-06-19:
I do not understand: the tunable was renamed to `elementAddDelay` with a deprecation warning on `elementStartDelay`. And the code properly handle `elementAddDelay` as `elementStartDelay`. 

What is actually not working with the `elementAddDelay` tunable?

jerome-benoit, 2024-07-02:
Will be closed soon unless the reporter gives more information on the issue.

closed 2024-07-03 by jerome-benoit