SAP/e-mobility-charging-stations-simulator #1103: [BUG] Request PDU is invalid

saschnet, 2024-07-22:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

Hi Jerome,

I just tried to connect the simulator with our backend and experienced two errors.

**1. OCPP 2.0.1 "StatusNotification"**

Sending the OCPP 2.0.1 Receiving the StatusNotification response, e.g. with the following contents for example:
`[3, 'f17dfb52-e345-43fa-b3ae-751946f9c5c7', {'customData': null, 'currentTime': '2024-07-22T18:56:23.498588Z', 'interval': 15, 'status': 'Accepted', 'statusInfo': null}]`

yields the following error:
```
error: 22.7.2024, 20:49:32 CS-KEBA-00001 | ErrorUtils.handleSendMessageError: Send request command 'StatusNotification' error: Request PDU is invalid {"code":"FormatViolation","command":"StatusNotification","date":"2024-07-22T18:49:32.216Z","name":"T","stack":"T: Request PDU is invalid
at xr.validateRequestPayload (file:///dist/ChargingStationWorker.js:6:69877)
at xr.buildMessageToSend (file:///dist/ChargingStationWorker.js:6:72709)
at file:///dist/ChargingStationWorker.js:6:71646
at new Promise (<anonymous>)
at xr.internalSendMessage (file:///dist/ChargingStationWorker.js:6:70903)
at xr.sendMessage (file:///dist/ChargingStationWorker.js:6:69341)
at xr.requestHandler (file:///dist/ChargingStationWorker.js:6:92846)
at Pe (file:////dist/ChargingStationWorker.js:6:14018)
at n.startMessageSequence (file:///dist/ChargingStationWorker.js:7:29993)
at n.<anonymous> (file:///dist/ChargingStationWorker.js:6:114585)"}
```
The same error occurs using the ocpp-server from the test directory. I could not find any unit tests for this specific example.

For OCPP version 1.6 this is working - it is only affecting version 2.0.1.

**2. OCPP 1.6 "Authorize"**

For the OCPP version 1.6 the simulator works until the first authorization is attempted. Loading the response is also failing. Example:

The simulator publishes
`[2, '53c8b8f1-c43a-4f6f-a6d9-f25a14fdfe93', 'Authorize', {'idTag': 'UNDEFINED'}]`

Our backend responds
`[3, '53c8b8f1-c43a-4f6f-a6d9-f25a14fdfe93', {'idTagInfo': {'expiryDate': '2024-08-19T19:07:08.374410Z', 'parentIdTag': null, 'status': 'Invalid'}}]`

Error in the logs:
```
error: 22.7.2024, 21:07:08 CS-KEBA-00001 | OCPP16ResponseService.responseHandler: Handle response error: Response PDU is invalid {"code":"FormatViolation","command":"Authorize","date":"2024-07-22T19:07:08.159Z","name":"T","stack":"T: Response PDU is invalid
at wr.validateResponsePayload (file:///dist/ChargingStationWorker.js:6:76113)
at wr.validatePayload (file:///dist/ChargingStationWorker.js:6:82157)
at wr.responseHandler (file:///dist/ChargingStationWorker.js:6:81606)
at g (file:///dist/ChargingStationWorker.js:6:70116)
at o.handleResponseMessage (file:///dist/ChargingStationWorker.js:7:25797)
at o.onMessage (file:///dist/ChargingStationWorker.js:7:26310)
at WebSocket.<anonymous> (file:///dist/ChargingStationWorker.js:7:4320)
at WebSocket.emit (node:events:517:28)
at Receiver.receiverOnMessage (node_modules\\.pnpm\\ws@8.17.1_bufferutil@4.0.8_utf-8-validate@6.0.4\\node_modules\\ws\\lib\\websocket.js:1211:20)
at Receiver.emit (node:events:517:28)"}
```

This issue is independent on whether to send acceptance or rejections.

I could also not find any unit tests for this specific example.

Both errors contain invalid PDU. It might make sense to test all potential answers. If helpful, I could provide some backend examples to you.

### Version

latest

### Node.js version

v18.20.2

### System

Unrelated

jerome-benoit, 2024-07-22:
> 
> Both errors contain invalid PDU. It might make sense to test all potential answers. If helpful, I could provide some backend examples to you.
> 

The CSMS is indeed sending invalid payload response. You can relax OCPP command payload check with the OCA JSON schema with a CS template tunable . 
But `null` as a value for a field is not a valid value, the CSMS is not properly building/serializing payloads for at least the command quoted in that issue.

saschnet, 2024-07-22:
Thanks a lot for the quick and helpful help.

Indeed the payload check is not very forgiving (which is very good!). I was actually not aware myself that there is a difference for the schema validation whether to set a variable null or to skip it completely. After changing settings that keys with nulls are skipped, the simulator is able to communicate correctly with the Authorize command.

However, the response to status notifications is still not working correctly.
What is wrong with the following response?

Call from the Simulator: `[2, '62c8f84f-9f17-4be3-899b-3a16ea5ba095', 'StatusNotification', {'connectorId': 1, 'status': 'Available', 'errorCode': 'NoError'}]`
Respone from the CMSM: `[3, '62c8f84f-9f17-4be3-899b-3a16ea5ba095', {}]`

The response is still rejected by `[Response PDU is invalid](error: Request PDU is invalid {"code":"FormatViolation","command":"StatusNotification","date":"2024-07-22T20:58:00.201Z","name":"T","stack":"T: Request PDU is invalid)`

[image]

saschnet, 2024-07-24:
@jerome-benoit To avoid that you might have overseen my question as one part seems like a real bug to me:

Is anything wrong with the following response or does the parser have issues with empty payloads (its working fine for the heartbeat response, though)?

Call from the Simulator: `[2, '62c8f84f-9f17-4be3-899b-3a16ea5ba095', 'StatusNotification', {'connectorId': 1, 'status': 'Available', 'errorCode': 'NoError'}]`
Respone from the CMSM: `[3, '62c8f84f-9f17-4be3-899b-3a16ea5ba095', {}]`

The response is still rejected by ```[Response PDU is invalid](error: Request PDU is invalid {"code":"FormatViolation","command":"StatusNotification","date":"2024-07-22T20:58:00.201Z","name":"T","stack":"T: Request PDU is invalid)```

jerome-benoit, 2024-07-24:
Won't fix and works with: 

- https://github.com/steve-community/steve
- https://github.com/mobilityhouse/ocpp
- SAP e-Mobility OCPP server
- ... 

=> CSMS bug

closed 2024-07-24 by jerome-benoit

saschnet, 2024-07-24:
It does not work with the test server you provide yourself in the tests directory which is related on the link you mention yourself: https://github.com/mobilityhouse/ocpp.

Steps to reproduce: 
1. Start OCPP Server in test directory
2. Start and connect via master branch.
3. Look into the logfiles and find the error.

=> e-mobility-charging-stations-simulator bug

jerome-benoit, 2024-07-24:
> It does not work with the test server you provide yourself in the tests directory which is related on the link you mention yourself: https://github.com/mobilityhouse/ocpp.

It's an OCPP2 only mock server and the simulator OCPP2 stack is not supporting the `StatusNotification` command yet. 

OCPP 1.6 server using https://github.com/mobilityhouse/ocpp just works. 

When you have critical and noticeable issue with a software that has a relatively large users base and you are the first to encounter it, the issue is not on the software side. Otherwise it would have been reported since. 

Please learn how to debug network protocol using HTTP.

saschnet, 2024-07-24:
> It's an OCPP2 only mock server and the simulator OCPP2 stack is not supporting the `StatusNotification` command yet.

The readme claims it is supported:
[image]

> OCPP 1.6 server using https://github.com/mobilityhouse/ocpp just works.
> 
> When you have critical and noticeable issue with a software that has a relatively large users base and you are the first to encounter it, the issue is not on the software side. Otherwise it would have been reported since.

Since the OCPP2 commands are limited I doubt that it has a noticable user base for this version. Due to the lack of unit tests for this use case, be better safe than sorry - if nobody reports anything the project cannot improve. Please trust me, that I have not submitted this issue after the first connection, but spent several hours trying to find the issue on my side. Since I have not found it and the "solutions" from this thread have been contradictory, I stayed persistant for the sake of a stable product.

> Please learn how to debug network protocol using HTTP.

For the sake of a positive environment in the community, I recommend to provide your community to tell them what kind of debugging information you would have expected rather than simply stating that someone is not capable in debugging.

jerome-benoit, 2024-07-24:
The OCPP2 `StatusNotification` is working as intended with a CS template defining EVSEs. 
The issue is lying with unclear error message at payload validation. But it's easy to compare the sent payload with the JSON schema to find out the culprit field.

saschnet, 2024-07-24:
Thank you very much. Now it makes absolutely sense!