SAP/e-mobility-charging-stations-simulator #1227: [BUG] StatusNotification Available sent after RemoteStopTransaction -> StopTransaction.

entropical, 2024-11-25:
### Duplicate issue

- [X] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

I am new to this simulator. I got it working, and it definitely seems promising.

The message flow is:

1. OCPP Server: RemoteStopTransaction
2. ChargePoint Simulator: StopTransaction
3. ChargePointSimulator: **StatusNotification Available**.

That StatusNotification Available is sent automatically, but this does not happen with the actual chargers we are using (ABB). StatusNotification is only sent when the physical connector is unplugged.

Logs from my server:
```
Nov 25, 2024 @ 12:34:21.604	RAW_RCV: [2, "875d437e-63c6-4c7f-a52b-a0184a4ffe7b", "StatusNotification", {"connectorId"=>1, "errorCode"=>"NoError", "status"=>"Available"}]
Nov 25, 2024 @ 12:34:21.516	RAW_SND: [3,"80226546-d258-4e4f-acff-a3dfbb1fa226",{"idTagInfo":{"status":"Accepted"}}]
Nov 25, 2024 @ 12:34:21.503	RAW_RCV: [2, "80226546-d258-4e4f-acff-a3dfbb1fa226", "StopTransaction", {"idTag"=>"VID:999999999985", "meterStop"=>0, "timestamp"=>"2024-11-25T01:34:21.493Z", "transactionId"=>8346361, "reason"=>"Remote"}]
Nov 25, 2024 @ 12:34:21.471	RAW_SND: [3,"adcb7a93-f27a-4c8a-928e-35e7eecb85e7",{}]
Nov 25, 2024 @ 12:34:21.440	RAW_RCV: [2, "adcb7a93-f27a-4c8a-928e-35e7eecb85e7", "StatusNotification", {"connectorId"=>1, "errorCode"=>"NoError", "status"=>"Finishing"}]
Nov 25, 2024 @ 12:34:21.414	RAW_RCV: [3, "0012cc24-d8a5-4d1e-ad37-cf42a9acdcfb", {"status"=>"Accepted"}]
Nov 25, 2024 @ 12:34:21.235	RAW_SND: [2,"0012cc24-d8a5-4d1e-ad37-cf42a9acdcfb","RemoteStopTransaction",{"transactionId":8346361}]
```

### Version

latest

### Node.js version

v22.3.0

### System

  System:
    OS: Linux 6.8 Ubuntu 22.04.5 LTS 22.04.5 LTS (Jammy Jellyfish)
    CPU: (12) x64 AMD Ryzen 5 7530U with Radeon Graphics
    Memory: 6.00 GB / 14.46 GB
    Container: Yes
    Shell: 5.1.16 - /bin/bash

### Expected result

StatusNotification Available should NOT be sent automatically following a StopTransaction. 

StopTransaction may be sent as a result of the RemoteStopTransaction, or the Stop button on the HMI on the charger. 

### Actual result

Simulator sent Status Notification Available

jerome-benoit, 2024-11-25:
It's the intended behavior since there's no simulation at the connectors level: block/unblock, plug/unplug.

entropical, 2024-11-25:
Fair enough. My use case involved potentially multiple remotestart/remotestop commands, which needs the connector to stay connected. 

Tool works well in general though, thanks for your work.