SAP/e-mobility-charging-stations-simulator #1299: [BUG] The requested module 'node:crypto' does not provide an export named 'hash'

akashkpfreelancer, 2025-01-27:
### Duplicate issue

- [x] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

SyntaxError: The requested module 'node:crypto' does not provide an export named 'hash'       
    at ModuleJob._instantiate (node:internal/modules/esm/module_job:132:21)
    at async ModuleJob.run (node:internal/modules/esm/module_job:214:5)
    at async ModuleLoader.import (node:internal/modules/esm/loader:329:24)
    at async loadESM (node:internal/process/esm_loader:34:7)
    at async handleMainPromise (node:internal/modules/run_main:113:12)

Node.js v20.10.0
 ELIFECYCLE  Command failed with exit code 1.

### Version

latest

### Node.js version

20.10.0

### System

Please fix this bug

### Attachments

[image]

jerome-benoit, 2025-01-28:
Please use the latest current node LTS version. 
The version check in package.json should be refined.

closed 2025-04-01 by jerome-benoit

jerome-benoit, 2025-04-01:
Should be fixed by https://github.com/SAP/e-mobility-charging-stations-simulator/commit/678ffc93c87c3c472679ea92b2732d2a7d0513f6