SAP/e-mobility-charging-stations-simulator #1364: [BUG] Get Configuration with empty parameter list returns empty result

LisaLamparterXX, 2025-03-28:
### Duplicate issue

- [x] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

According to the OCPP 1.6 protocol the "Get Configuration" request should return a list of all configuration settings if the parameter is empty or missing.
In the current implementation, an empty parameter list return an empty result.

The respective code is in `OCPP16IncomingRequestService.ts` in the method `handleRequestGetConfiguration` (line 1042). Instead of `if (key == null)` the following would fix it `if (!key || key.length === 0)`.

### Version

2.0.2

### Node.js version

node:lts-alpine

### System

Tested with docker container based on node:lts-alpine

### Expected result

GetConfiguration with empty string list parameter should behave like if the parameter is null and return the list of all available configuration settings.

### Actual result

GetConfiguration with empty string list parameter returns a result with the field `configurationKey` and `unknownKey` being empty arrays.

jerome-benoit, 2025-03-28:
Hello,

Would you do the PR that addresses the issue?

LisaLamparterXX, 2025-03-28:
Hi, 

yes, I can provide a pull request 👍

commit d13819a6 (2025-04-01): fix: properly handle empty key array in GetConfiguration (#1368)

closes #1364
PR #1368 (2025-04-01): fix: properly handle empty key array in GetConfiguration

## PR Checklist

- [x] Please add a description of your changes.
- [x] Please ensure title follows [conventional commits](https://www.conventionalcommits.org/en/v1.0.0/).
- [ ] We need your changes to come with unit tests in order to keep this project in quality and easy to maintain.
- [ ] If your changes contain any new feature please be sure to document it.
- [x] Please add a link to the open issue or task that this pull request will resolve.

## Does this PR introduce a breaking change?

- [ ] Yes
- [x] No

If yes please indicate it  [... cut]
closed 2025-04-01 by jerome-benoit