SAP/e-mobility-charging-stations-simulator #1374: [BUG] Missing retry on boot notification response 'pending'

LisaLamparterXX, 2025-04-03:
### Duplicate issue

- [x] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

As far as we understand the OCPP 1.6 specification, the charging point should perform automatic retries if the central system answers with "Pending" to the BootNotification and the "interval" in the BootNotification response shall be used as retry interval.

The ocpp protocol says 

> If the Central System returns
something other than Accepted, the value of the interval field indicates the minimum wait time before sending a
next BootNotification request. If that interval value is zero, the Charge Point chooses a waiting interval on its
own, in a way that avoids flooding the Central System with requests

Currently, the simulator does not perform any retry if the central system responds with Pending.
Our tests showed, that it would work if the following two code snippets are adapted:

* ChargingStation.ts, method "onOpen", line 2112 and line 2131: Condition should be something like `this.inUnknownState() || !this.inAcceptedState()`
* OCPPRequestService.ts, methode "internalSendMessage", line 246:  first condition should be something like `((chargingStation.inUnknownState() || chargingStation.inPendingState()) && commandName === RequestCommand.BOOT_NOTIFICATION)`

Any chance this can be adapted or are we misinterpreting the ocpp protocol here?
Thanks in advance!

### Version

2.0.4

### Node.js version

node:lts-alpine

### System

Tested with docker container based on node:lts-alpine

### Expected result

The simulated charging points perform automatic retries when central systems reponds with "pending" to the boot notification. The "interval" from the boot notification response should be used as retry interval.

### Actual result

If the central systems responds with "pending" to the boot notification, the simulator doesn't send a boot notification again for this charging point.

### Steps to reproduce

* Start charging point
* let the central system answer with "pending" to the boot notification
* the simulators does not send a boot notification again

jerome-benoit, 2025-04-03:
Have you checked the OCPP errata? The retry logic at boot notification is debatable in 1.6 version. 

For example, too aggressive to allow a CSMS with thousands of CS to reconnect gracefully after a crash.

LisaLamparterXX, 2025-04-04:
Thanks for the hint. I've checked the OCPP 1.6 errata again and I could not find the respective part that refers to boot notifications responses with 'pending'.

The scenario in our case is that we want to change some configuration values before we accept a charging point and therefore respond with 'pending' until we are finished to do so.
Do we have to assume that the charging point does not send a boot notification request again by its own? So we should trigger a boot notification from the central system instead?

jerome-benoit, 2025-04-04:
Since the condition  checked is: 

```ts
 public isRegistered (): boolean {
    return !this.inUnknownState() && (this.inAcceptedState() || this.inPendingState())
  }
```

I cannot reproduce the issue: boot notification is retrying if the CS is in pending state.

LisaLamparterXX, 2025-04-06:
Unfortunately, the boot notification retry logic is only applied, if the charging station is considered as "not registered" (see ChargingStation.ts., method `onOpen()`). And according to the `isRegistered()`-method you've pasted above, a charging station in pending state is considered as "registered".
Therefore, the retry logic is only applied, if the state is unkown or rejected.

jerome-benoit, 2025-04-07:
Could you please cross-test the associated PR before its merge?

LisaLamparterXX, 2025-04-07:
Yes, no problem. I will test the associated PR.

LisaLamparterXX, 2025-04-08:
Hi @jerome-benoit ,

thanks for adapting it so fast. I've tested the PR in our environment and it worked as expected👍 
However, I was only able to test the OCPP 1.6 part.

PR #1376 (2025-04-08): fix: ensure CS in pending state can send boot notification

## PR Checklist

- [x] Please add a description of your changes.
- [x] Please ensure title follows [conventional commits](https://www.conventionalcommits.org/en/v1.0.0/).
- [ ] We need your changes to come with unit tests in order to keep this project in quality and easy to maintain.
- [ ] If your changes contain any new feature please be sure to document it.
- [x] Please add a link to the open issue or task that this pull request will resolve.

## Does this PR introduce a breaking change?

- [ ] Yes
- [x] No

If yes please indicate i [... cut]
commit b72d1749 (2025-04-08): fix: ensure CS in pending state can send boot notification (#1376)

* fix: ensure CS in pending state can send boot notification

closes #1374

* fix: proper CS state conditions

---------