SAP/e-mobility-charging-stations-simulator #1475: [FEATURE] Authorize ID Tags when starting new transactions

PatrickDale, 2025-07-23:
### Duplicate issue

- [x] I've searched for any related issues and avoided creating a duplicate issue.

### Component

Simulator

### Description

When the ATG starts a new transaction, it [authorizes the ID tag of the transaction](https://github.com/SAP/e-mobility-charging-stations-simulator/blob/main/src/charging-station/AutomaticTransactionGenerator.ts#L456) against the OCPP-J server when the `requireAuthorize` ATG option is set to `true`.

When starting a new transaction without the ATG though, the ID tag is not authorized against the OCPP-J server.

Having the ability to authorize ID tags against the OCPP-J server for non ATG initiated transactions would be very valuable for testing transaction authorization.

I believe this functionality could be implemented by reusing the [isIdTagAuthorized](https://github.com/SAP/e-mobility-charging-stations-simulator/blob/main/src/charging-station/ocpp/OCPPServiceUtils.ts#L111) function when [handling StartTransaction requests](https://github.com/SAP/e-mobility-charging-stations-simulator/blob/main/src/charging-station/ocpp/1.6/OCPP16RequestService.ts#L159).

jerome-benoit, 2025-07-23:
I fail to understand the feature request: using the web ui or the ui service with curl,... allows to do a start transaction with any user defined payload, including defining the idtag.

PatrickDale, 2025-07-23:
The feature request is to authorize the idtag against the OCPP backend when starting a transaction.

The simulator does this for ATG initiated transactions, but does not do this for non ATG initiated transactions.

PatrickDale, 2025-07-23:
@jerome-benoit maybe I should have opened a bug issue. The feature that is missing though is that when you start a transaction via the UI, the idtag is not authorized against the OCPP backend. Authorization logic is in place though for transactions started via ATG.

I opened a PR to address this using existing simulator functionality: https://github.com/SAP/e-mobility-charging-stations-simulator/pull/1476.